FYSETC/FYSETC-SPIDER #150: Spider v2.3; USB problems; no MCU id.

Fickert07, 2024-01-31:
Able to flash bootloader and firmware via USB with STM32 Programmer, additionally able to flash firmware Via SD (confirmed with old.bin file on sd card after install), but Pi does not detect a USB change with sudo dmesg when I unplug and plug back in the board. 

Pi is not the issue as it is detecting multiple Webcams via USB. 

Not sure where to go from here other than order a new board. Frustrating as I purchased a year ago, but was never taken out of the plastic. 

Any tips I will be greatful.

Fickert07, 2024-01-31:
Also to add, firmware compiled was using USB not serial or CAN.

zhujue888, 2024-02-04:
看来没有得到任何回复。

michaelcuneo, 2024-09-18:
I have the exact same problem, in USB DFU mode the firmware can be flashed, then it finishes flashing, drops out of DFU Mode to detect the USB and I have a working printer... as soon as I pull the Jumpers off U5V and 3.3V and restart, USB is gone.

Eloura74, 2025-05-12:
same problem*