FYSETC/FYSETC-SPIDER #155: Spider 3 with TMC2209 V4.0 (fysetc) stepper drivers

ozzman39, 2024-04-13:
TMC2209 V4.0

ozzman39, 2024-04-20:
Just tried switching the stepper in that slot thinking maybe I had a bad one same error.

ozzman39, 2024-04-21:
what are the chances of 2 stepper driver slots being dead on a board? getting  (Unable to read tmc uart 'stepper_(X andY)' register IFCNT)  if i move the steppers over to the end of the board it I can clear the error (s). the pinout looks correct (github and photo) Spider 3 
[image]

eliwong, 2024-04-22:
[image]
I'm very sorry that the video is wrong and bothered you. The power supply next to XY is prepared for 48V. If you do not use 48V, you need to short-circuit 24V and 48V so that the XY socket will have power.
Both 2209 V3.1 and V4.0 can be used on PSIDER 3.0. All it takes is a jumper cap.

ozzman39, 2024-04-22:
Thank you so much for responding! could you send a picture to illustrate
please.

ozzman39, 2024-04-22:
Saw the pick figured it out thank you so much!!!!!!

lerototi, 2024-12-05:
Hey Guys! I'm having the same problem. 
Can you say how I solve it? 
I'm using spider 2.3 and tmc 2209 v4.0 by fysetc kit. I don't understanding what I need to do to solve it.

TheAsda, 2024-12-06:
Today i've also encountered while upgrading from spider v2.2 to v3.0. In my case the issue was that board was prepared for different drivers. I've used TMC 2209 v3.1 from my old board. So i removed the unnecessary jumpers and it worked.
[image]

lerototi, 2024-12-06:
Nice!!! It worked for me!! Thanks!!