FYSETC/FYSETC-SPIDER #158: Spider v3 cant run BTT EBB 2209 CAN toolhead (Fysetc Voron 2.4 R2)

inkvi601, 2024-07-09:
Id like to ask you, if there is any compatibility problem to use BTT EBB 2209 CAN toolhead with Spider V3 controller?

I flashed firmware on both (Spider3 and EBB CAN module), controller is normally running using CAN bus, but when I try to turn on the EBB CAN toolhead module, entire electronics is silent - there is only one red LED flashing (24V label red LED on Spider).

I dont understand, where can be a problem, since the toolhead module is using same voltage. Everything is working with original Fysetc CAN toolhead module... Im using original FYSETC CAN cable provided with the kit.

inkvi601, 2024-07-10:
I found the issue. You have error in your drawing. 

Your original wiring image from Fysetc wiki: https://wiki.fysetc.com/assets/SB_CAN_ToolHead_V13_Wiring.png

You have +24V (red cabel) on the bottom pin fo CAN cable connector. But in reality, you have it on the top pin (you have it right on Voron wiring image). So you have it swapped- drawing from wiki vs. reality.

[image]

barandarenato, 2025-11-05:
@inkvi601 After fixing the cables position, were you able to connect Spider V3 with the BTT EBB 2209 CAN?
I'm looking into this alternative to get away from the Voron Tap and use the BTT Eddy instead.

inkvi601, 2025-11-06:
Yes its working. I have two Vorons running Spider V3 (second is newer FYSETC Spider V3.0 H7) MCU's with BTT EBB 2209 CAN for more than 1 year without any issues.

cyberjak2k, 2025-11-06:
I previously had Spider 3 , and recently changed to the H7 , it was an easy transition and continues to work with my ebb2209 CANbus toolhead. 
Firstly your initial post suggested wiring incorrect. 
I was curious however as you linked a non working link with a PNG to Fysetc wiki. Yet the toolhead is a Bigtreetech. This is when colors should NOT be observed and go with understanding of pins to related pins , including but not limited to power. 
What I'm not clear on is have you tried using Esoterical guide to Canbus to guide you step by step?
https://CANbus.esoterical.online
I've followed this as guidance every time , to double check myself on getting things running. But wiring will have to be based on your own knowledge of understanding not just wiring guide but the pin documents which help identifying where the connection is on the board. 
If you get stuck. Find me in discord if you want