FYSETC/FYSETC-SPIDER #162: Spider H7 repo does not exist

cereal7802, 2024-08-27:
I bought the Spider-H7 and on the packaging it shows a link to a wiki article that does not exist, and a github repo that does not exist.

Links:

https://wiki.fysetc.com/FYSETC-SPIDER-H7
https://github.com/FYSETC/FYSETC-SPIDER-H7

it would be nice if you would make those resources public before shipping products to customers....

wickated, 2024-09-02:
u actually dont need repo since all the pins are written directly on board face

grc1000, 2024-09-10:
Hello, is there no example for configuring printer.cfg in klipper?

cyberjak2k, 2024-09-10:
Please see the readme section below the folders. Printer.cfg is there but
examples for klipper  are usually based on printer
https://github.com/FYSETC/FYSETC-SPIDER/tree/main/firmware/Klipper

grc1000, 2024-09-19:
Hello, my board arrived, but I can't find documentation regarding the spider h7, only the f446 version, which is very different. I can't write the firmware to the SD or put the board in dfu boot mode. Where can I find the correct documentation for my spsder h7 model?

Rushmere3D, 2026-02-04:
OP's issue resolved.

closed 2026-02-04 by Rushmere3D