FYSETC/FYSETC-SPIDER #167: The Spider v3.0 board doesn't communicate with the BTT-PI board via the UART port.

solon-nascimento, 2025-07-18:
I bought a Spider v3.0 and a BTT-PI to use with my new Voron Trident. I installed Klipper on the board according to Bigtreetech's recommendations, as per the link https://github.com/bigtreetech/CB1. I changed the Boardenv.txt file to UART communication (console=serial). I used the firmware in the bootloader from the link https://github.com/FYSETC/FYSETC-SPIDER/tree/main/firmware/Klipper and loaded it via SDCARD. In printer.cfg, I configured it (serial: /dev/ttyAMA0), and even after all these changes, the connection between the BTT-pi and the Spider board doesn't work. I need support to get it working.

[image]

[image]

[image]

[image]

solon-nascimento, 2025-07-19:
Thanks for the quick reply. The file already has console:serial as shown in the image below, so that wouldn't be the problem. When I changed serial: /dev/ttyAMA0 to serial: /dev/ttyS0, Klipper gave me this version error. I don't know what to do anymore; I've been trying to get it to work for three days. I also can't find anything specific about using these two cards together. I also apologize for my English; Google is helping me with this!

[image]

[image]

cyberjak2k, 2025-07-19:
Yes that's because the Klipper version on the CB1 is different than the one on the stm32 of the Mainboard. This is why I use the USB not the UART as it makes much easier to flash new firmware when required. Eliminating the need of SDcard flashing but in short remake the Klipper binary from Klipper on your CB1 and reflash the mainboard with the resulting bin file. And reconsider moving to usb so you can then also take advantage of the usb to can bridge should you wish to do as I did and use CANbus to connect to the toolhead

solon-nascimento, 2025-07-19:
I'll be doing it now

solon-nascimento, 2025-07-19:
I followed your advice and just compiled the firmware using DFU communication. Everything seemed fine, but it gave me an error displaying the USB port communication code. I'll copy it below. If you could analyze it and tell me if I did something wrong so I can finally get everything working, I'd be very grateful.

  Preprocessing out/src/generic/armcm_link.ld
  Linking out/klipper.elf
  Creating hex file out/klipper.bin
biqu@VORON:~/klipper$ lsusb
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 0483:df11 STMicroelectronics STM Device in DFU Mode
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
biqu@VORON:~/klipper$ make flash FLASH_DEVICE=0483:df11
  Building hid-flash
gcc -c -Wall -std=gnu99 -I . `pkg-config libusb-1.0 --cflags` main.c -o main.o
gcc -c -Wall -std=gnu99 -I . `pkg-config libusb-1.0 --cflags` hid-libusb.c -o hid-libusb.o
gcc -c -Wall -std=gnu99 -I . `pkg-config libusb-1.0 --cflags` rs232.c -o rs232.o
gcc -no-pie main.o hid-libusb.o rs232.o `pkg-config libusb-1.0 --libs` -lrt -lpthread -o hid-flash
  Flashing out/klipper.bin to 0483:df11
sudo dfu-util -d ,0483:df11 -R -a 0 -s 0x8000000:leave -D out/klipper.bin

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Downloading to address = 0x08000000, size = 29000
Download        [=========================] 100%        29000 bytes
Download done.
File downloaded successfully
dfu-util: Error during download get_status

Failed to flash to 0483:df11: Error running dfu-util

If the device is already in bootloader mode it can be flashed with the
following command:
  make flash FLASH_DEVICE=0483:df11
  OR
  make flash FLASH_DEVICE=1209:beba

If attempting to flash via 3.3V serial, then use:
  make serialflash FLASH_DEVICE=0483:df11

make: *** [src/stm32/Makefile:111: flash] Error 255
biqu@VORON:~/klipper$ ls/dev/serial/by~id
-bash: ls/dev/serial/by~id: No such file or directory

closed 2026-02-04 by Rushmere3D